ArduPilot/MissionPlanner #1895: Mission Planner: no option for onboard mag cal with wifi connection

kd0aij, 2018-07-19:
### Bug report

only live cal available with wifi, only onboard when connected via USB

**Version**
master
[image]
[image]

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad X

**Hardware type**
Pixracer/ESP8266 with external I2C compass/GPS

**Logs**
_Please provide a link to any relevant logs that show the issue_

meee1, 2018-07-19:
how uptodate is your esp8266? this is likerly the esp8266 nothing to do with missionplanner

kd0aij, 2018-07-19:
Interesting. Does the ESP8266 need to understand mavlink?