ArduPilot/MissionPlanner #1898: UAVCAN MAVLink message handling

olliw42, 2018-07-20:
Hey Michael,

I did ask about that topic before in https://discuss.ardupilot.org/t/missionplanner-capable-of-handling-uavcan-mavlink-messages/21038, but thought I would better come over to here. Hope that's ok, and hope it's ok to discuss some more points.

I now indeed did the first steps towards adding UAVCAN mavlink message support to my AP fork, especially it sends out the UAVCAN_NODE_STATUS message, and MP indeed recognizes it and lists an additional component, and would let me allow to retrieve the parameters (which I have not yet implemented). Fantastic, THX for this addition to MP.

However, I also emit the UAVCAN_NODE_INFO message when the UAVCAN node gets detected as online, but I couldn't find MP reacting in any way to it. This might be because I'm doing something wrong, or because MP indeed isn't supporting it.

So, question, pl: 
Does MP digest UAVCAN_NODE_INFO, and if so, how would I be able to notice that?

E.g., I would think that a good place to show the received node name in the info field below the COM and BAUD rate fields (maybe this field should be made a bit wider then).

Question pl:
Does MP emit MAV_CMD_UAVCAN_GET_NODE_INFO messages, and if so, on which conditions?

I'm sure it is in MP, but I couldn't find an in-situ method to check what MAVLink messages MP is receiving and sending (the buttuon in the ctrl-F window I couldn't get to do anything). What would be the recommended way to spy the MAVLink communication in MP?

Thx so much for all the efforts you put into MP,
Olli

meee1, 2018-07-25:
to spy on inbound packets, control-f > mavlink inspector.
MP currently does nothing with UAVCAN_NODE_INFO, and is not decoding it internally anywhere
using the mavlink inspector you should be able to see the content of the packet in real time

sediq-khan, 2019-02-11:
Hi,

I need to add a new message to MAVLink message linst and display it on the Mission Planner, could any one assist me in that .i.e how can I add a message to MAVLinks message list and how to display it on the Mission Planner.

Thank you

olliw42, 2019-05-15:
can be closed I think

closed 2019-05-15 by olliw42