ArduPilot/MissionPlanner #1900: Hard coded "config directory" (into windows documents folder) causing issues on synced computers

pompecukor, 2018-07-22:
#### Issue details
Non user selectable config directory in "documents" folder casues trounble on PC/laptop with document sync running.

#### Version
Any

#### Platform
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

The problem I have is that I have my entire Windows "Documents" folder continuously synced between my Desktop PC and my laptop. I run individual copies of MP on both, but unfortunately, because it seems hard coded to use documents folder. It causes a lot of issues. Starting from simple things like if I resize a window in the laptop, then when I run MP on the desktop is gets all messed up. And there are more serious issue too, for example when one gets updated and I try to update the other and many more. All because the config files are forced to ".../documents/Mission Planner folder". I would love to sort this out somehow.
This is not same as log, which you can easily change from the setting. Would be awesome to have something similar to that option. So that user could specify another location for cofig stuff. Obviously this would be needed at the installation phase. 
Thank you.

pompecukor, 2018-07-22:
[image]

meee1, 2018-07-25:
exclude the folder from your syncing app is the easiest fix

pompecukor, 2018-07-26:
Good idea, I completely forgot that overdrive let you remove certain sub-folder from a sync folder.
Thank you.