ArduPilot/MissionPlanner #1901: Clarify Set-Home-Here and Set-EKF-Origin

rmackay9, 2018-07-23:
ArduPilot allows the user to set a "Home" and/or an "EKF Origin" position.

1. users who want their vehicle to always return to a specific location (regardless of where it was armed) will want to use the "Set Home Here" option.
2. users of vehicles without a GPS must initialise the EKF by telling it where it is in the world.  These users must use the "Set EKF Origin" option.

In addition to the above, I believe the MP maintains it's own copy of home which it uses to center the Flight Plan screen's map and also which it sends to the vehicle as the first mission item (which is ignored actually but it needs to be there).

It is not clear how users can do item 1 (i.e. "Set Home Here").  The Flight Data screen has a right-mouse-button menu option to "Set Home Here >> Set EKF Origin" but I believe this is item 2.. the naming is odd and confusing because it's nesting the "Set EKF Origin" under "Set Home Here".  I'm also not sure why it's nested when there's only one option.

I guess I'm confused how a user is meant to set the home position.  This is for both Copter and Rover.

meee1, 2018-07-25:
both 1 and 2 exist in flightdata. as both options are valid, ie Set Home Here if clicked will send a do_set_home

rmackay9, 2018-12-04:
We can close this now, fixed long ago

closed 2018-12-04 by rmackay9

maroquio, 2019-09-30:
Hello.
Does anyone know how I can set EKF origin using a dronekit/mavlink message?

vusaalab, 2019-10-23:
Try this one https://github.com/hoangthien94/vision_to_mavros/blob/master/scripts/set_origin.py

cnpcshangbo, 2020-08-25:
Hi, Randy @rmackay9 I have a recent indoor flight test that I used QGroundControl and forgot to set EKF origin.
I later found that the drone failed to stop when the GUIDED vx set to 0. I also found that the DF log doesn't have POS log. Do you think the velocity behavior is related to the absence of EKF origin?
[image]

rmackay9, 2020-08-25:
@cnpcshangbo,

If the velocity controller is running I think it should stop regardless of whether the EKF origin has been set or not.  I'd probably need to see a log though.  I'm not sure this issue is the best place to discuss this.  I think either a discussion in the support forums or on the discord "vision projects" channel would be better.  Send me a link when you decide where to post.