ArduPilot/MissionPlanner #1903: Copter Extended Tuning page updates for Copter-3.6 release

rmackay9, 2018-07-23:
A few parameters have changed names in Copter-3.6 so the extended tuning page needs to be updated:

- ACCEL_Z_* renamed to PSC_ACCZ_*
- ACC* parameters renamed to PSC_ACC*
- POS* parameters renamed to PSC_POS*
- VEL* parameters renamed to PSC_VEL*

[image]

[This issue was brought up by a beta tester in this discussion](https://discuss.ardupilot.org/t/extended-tuning-parameters-to-zero-after-firmware-update-arducopter-3-6rc6/31070/2).

sq7lrx, 2018-07-25:
also loiter speed is missing (readonly field)

rmackay9, 2018-07-26:
@sq7lrx txs for spotting that.

@meee1, so WPNAV_LOIT_SPEED has become just LOIT_SPEED now.

meee1, 2018-07-26:
yep, already fixed that

rmackay9, 2019-04-06:
all fixed so closing

closed 2019-04-06 by rmackay9