ArduPilot/MissionPlanner #1905: Bat monitor (second battery) "battery_usedmah2" value is nonsense.

pompecukor, 2018-07-23:
Bug report

We have 2 individual batteries in a setup. Both on their indivudial (mauch) sensors. Connected to port one and port two. Parameters are all identicall. Voltage and current valueas are correct, however battery_usedmah2 on the second battery is show is strange number. See screen capture below.
**Issue has been investigated on Mavlink/Arudpilot side and was found the data there was correct, so most probably Mission Planner bug.** 

See original issue on AP: https://github.com/ArduPilot/ardupilot/issues/8974#event-1748446363

Arduplane 3.10 latest beta
I do not think it is a version problem, we never tried it before and I dont think many people would either. As I said the other values are correct. This is a calculated value I believe. So hardware even is irrelevant.

Platform
[X] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

[image]

pompecukor, 2018-07-29:
@meee1 
Just a bit of update that might help iron this out, someone found an interesting work around:

1) With both batteries plugged in , then connect drone to mission planner, I get a huge number consumed on Batt2 Mah.
2) **With 1 batt plugged in first, then connect mission planner, then plug in Batt2, both Mah's work 100%**.

meee1, 2018-07-29:
I think this is already fixed in beta

pompecukor, 2018-07-29:
Epic, its working. Now, @meee1 would it be too much to ask to make the raw_temp (IMU temp) to have the decimal place corrected. That in degrees centigrade. So you need to move the decimal 2 places forward. I did not really want to open a new issue for this :)
I confuses the hell out of my users when I tell them to wait till IMU is up to 59C and they see a four digit number.

[image]