ArduPilot/MissionPlanner #1908: Time_in_air resets to zero after disarm

GrahamDyer, 2018-07-31:
#### Issue details
The "Time_in_air" value on the HUD resets to zero after disarming, leaving one to wonder how long the copter/plane/vtol actually flew unless you have a second person to make a note while you do the the landing, if you come back to the ground station this value (which counts up while flying) will have reset to zero. 

This never used to be like this, the value used to remain persistent until re-arming.

#### Version
v1.3.x

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Vtol & Copter

#### Hardware type
Pixhawk 2.1

#### Logs
N/A

meee1, 2018-07-31:
look for timeSinceArmInAir: 
and
timeInAir:

meee1, 2018-07-31:
make sure you are using the latest MP

closed 2018-07-31 by meee1