ArduPilot/MissionPlanner #1911: Motor Test: allow negative throttle % input

rmackay9, 2018-08-03:
It would be great if the Initial Setup >> Motor Test page allowed the user to provide a negative throttle percentage to indicate that they want to spin the motors in reverse. This might only be valid on Rovers and Boats although allowing it for all vehicles would probably be fine (it should be and if copters do something bad we could patch their motor test handling code).

The background for this request is recently while helping a developer with a skid-steering rover setup, we discovered that the relays used to control motor direction were reversed (i.e. left motor's relay was attached to flight controller's right-motor-relay control).  This incorrect setup could have been discovered more quickly if the Motor Test page supported providing a negative throttle percentage because then the user would have found that they couldn't make the motors spin in reverse.

rmackay9, 2019-06-11:
I just wanted to mention that this would still be useful to have.  For example [this user is having problems](https://discuss.ardupilot.org/t/skid-problems-v2/43465) getting the ESC setup correct on a rover and it would be helpful if he could test both forward and backwards.

rmackay9, 2019-06-12:
@meee1, great thanks!  [Here's another user support issue from yesterday](https://discuss.ardupilot.org/t/vehicle-does-not-drift-inside-of-loiter-radius/43362) that will benefit from this change... so thanks again

meee1, 2019-06-12:
@rmackay9 i did the change, but 100% untested in real life

rmackay9, 2020-10-29:
I've tested this and it works, thanks!

closed 2020-10-29 by rmackay9