ArduPilot/MissionPlanner #1913: RAD.RSSI and RAD.NOISE values recently broken

Naterater, 2018-08-03:
#### Issue details
I upgraded mission planner and no longer get RSSI or NOISE values displayed in the quick settings.

#### Version
1.3.57 and 1.3.57beta from 8/1/18
It worked fine in 1.3.56

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Traditional electric, BLHeli_32 ESC's

#### Hardware type
Cube Black

#### Logs
will dig up if requested.

meee1, 2018-08-12:
do you have a log you could upload? just to confirm

Naterater, 2018-08-12:
https://www.dropbox.com/sh/onnv9wxy602jxie/AADyrnNrbXaLPuVvg8dp-l0ba?dl=0

meee1, 2018-08-17:
what type of telem radio are you running?

Naterater, 2018-08-19:
RFD900X, and it is working again in beta. Thank you.

Naterater, 2019-12-04:
Pretty sure this broke again as I have witnessed the same issue after updating... https://discuss.ardupilot.org/t/rssi-missing-in-new-mission-planner-1-3-69/49870