ArduPilot/MissionPlanner #1916: SITL broken on master (1.3.58 build 1.3.6794.34444)

lvale, 2018-08-09:
Starting SITL via the GUI, no matter what vehicle, fails.

[image]

rmackay9, 2018-08-09:
I've noticed the MP simulator doesn't work as well right now.  It's unclear to me whether it's an ArduPilot change or an MP change that is to blame.  It's hit some teammates who are competing in a competition as well.. although they've found a way around the problem by starting the simulator manually.

lvale, 2018-08-09:
Starting the simulator from within MP and using QGC on the same machine works ok, so I guess that is a MP issue

lvale, 2018-08-09:
working now.
thanks @meee1 

closing

closed 2018-08-09 by lvale