ArduPilot/MissionPlanner #1917: MP window completely reconfigures itself on ARM command

Kelly-Foster, 2018-08-10:
Yesterday I upgraded MP to 1.3.58, build 1.3.6794.30567.  Running ArduPlane/ChibiOS latest (yesterday's build).  When I press the Arm/Disarm button on the Actions tab, the entire window shrinks to ~1/4 of its original size, font sizes shrink considerably.  Can't find a way to get it to revert back to full screen window and normal font.  MP remains usable, but tiny features.  Only workaround I have found is shut down and restart.  Windows 10.

Kelly

meee1, 2018-08-11:
please have a read here
https://www.howtogeek.com/175664/how-to-make-the-windows-desktop-work-well-on-high-dpi-displays-and-fix-blurry-fonts/

pretty sure the fix is the part down the bottom

DavidIngraham, 2018-09-13:
The most concerning thing about this is that it happens immediately on arming, which seems like a dangerous time for the GCS to change itself

DavidIngraham, 2018-09-29:
I can confirm that this is happening on low resolution displays as well as high resolution displays.

Kelly-Foster, 2018-09-30:
I was able to solve this behavior months ago, though I am rusty w.r.t. the solution.   I started by reading the link Michael posted above, and one of the posts talked about making an app-by-app change to thwart Windows behavior.  See this image of the per-app property that I recall solved the reconfig behavior.  Let us know if this works for you.  I vaguely recall having to find the MP app (and not a link to it) in order to change its attributes.

Kelly

[image]

Kelly-Foster, 2018-09-30:
Sorry, hit wrong reply button that closed issue.  Issue reopened.

Kelly