ArduPilot/MissionPlanner #1920: multiaircraft logfiles are put into wrong folder/s by first-seen-sysid, or something semi-random

davidbuzz, 2018-08-13:
#### Issue details
when MissionPlanner is connected to multiple aircraft, and gets packet/s for multiple different sysids, the behaviour that defines which folder the logfile/s are created appears to be pseudo-random.   

#### Version
all
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
not relevant
#### Hardware type
pixhawk, but any should be similar

#### Logs
look in User/Documents/Mission Planner/logs/FIXED_WING, and you'll see folder/s for each aircraft sysid, except for multi-vehicle ops with multiple sysids at once, it's not as you'd expect here.   you have to search all the subfolder/s for a logfile as it could be under any of the sysid/s that you were using at that time.

meee1, 2018-08-17:
thoughts on how this should be handled?
at the moment it will be random on one of the sysid's in the log

davidbuzz, 2018-08-18:
Ideas:
Lowest sysid in the log?  
Make a folder called 'multiple' and put it there?