ArduPilot/MissionPlanner #1924: Send Error message & cant escape  system null referenceException:Object reference not set to an instance of an Object

dwright361, 2018-08-17:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2018-08-17:
please fill in the form correctly.

closed 2018-08-17 by meee1

dwright361, 2018-08-17:
Version 1.3.57 build 1.3.6794.30567 ArduRover v3.4.2(318a941d)
Pixhawk

dwright361, 2018-08-17:
[image]