ArduPilot/MissionPlanner #1925: ESC Telemetry Display

Naterater, 2018-08-17:
#### Issue details
The newest releases of ardupilot are compatible with BL_Heli32 ESC telemetry. Individual ESC volts, current, and RPM can be logged and sent via telemetry, and it would be a huge benefit to be able to monitor this real-time rather than just looking at the averages and sums (BATT messages)

#### Version
Plane 3.9

#### Platform
[ X ] All
[  ] AntennaTracker
[ X ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Dual-motor plane

#### Hardware type
Cube Black. Lumineer BL_Heli32 51A ESC

#### Logs
Will post if requested.

meee1, 2018-08-17:
have a look for
esc1_volt... etc

closed 2018-08-17 by meee1

Naterater, 2018-08-17:
The current value shown is currently 100x the value that is logged. It's almost right, but not as of the latest beta from 1 hour ago.

meee1, 2018-08-18:
are they all x100? or just some fields?

Naterater, 2018-08-19:
Just some. Volt and Currnet are x100 while RPM is /10.

[image]