ArduPilot/MissionPlanner #1926: Pixhawk4 FMUv5 ArduRover firmware loading error

rg4444, 2018-08-20:
#### Issue details
Trying to upload Rover v 3.4.2 firmware to Pixhawk4 FMUv5 getting error as in picture attached.

#### Version
Using MissionPlanner 1.3.58 on WIn10
Windows got USB not recognized device message 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X] Rover
[  ] Sub

#### Airframe type
Rover

#### Hardware type
 Pixhawk4 FMUv5

#### Logs
Picture attached
[image]

meee1, 2018-08-20:
that version of rover is not available for the board version.

please try beta rover

rg4444, 2018-08-21:
Thank you! However here http://firmware.ardupilot.org/Rover/beta/PX4/ are only v4 versions. Any other place I could search for FMUv5 beta of ArduRover?

rg4444, 2018-08-21:
Got it. But none of them "Beta" or "latest"  doesn't have FMUv5 ...

rmackay9, 2018-08-21:
Ah right, sorry.  the fmuv5 is only supported under ChibiOS.  Here it is: http://firmware.ardupilot.org/Rover/latest/fmuv5/