ArduPilot/MissionPlanner #1928: SR1_XX parameters can not be changed in MP

hawaiianer, 2018-08-22:
I am trying to tune my crossfire diversity RX connection to MP via bluetooth because I have a lot of connection problems. Whenever I change any SR1 value and save, after a reboot the values stay like this:
SR1_ADSB 5
SR1_EXT_STAT 2
SR1_EXTRA1 4
SR1_EXTRA2 4
SR1_EXTRA3 2
SR1_PARAMS 10
SR1_POSITION 2
SR1_RAW_CT	5
SR1_RAW_SE 2
SR1_RC_CHAN 2

APM Copter Verion 3.5.7 ;Missionplanner: 1.3.58 @ Pixhawk 2 quad frame

If I change the values in APM Planner 2.0.24 they stay changed until I load up MP, then the values are back to the ones above.

Is this a bug in MP or am I missing something? In the manual it is mentioned that some values will be changed by the ground station=crossfire module, could this be the issue?

BR,
Lorenz

meee1, 2018-08-22:
use the stream rate settings in mission planner settings.

hawaiianer, 2018-08-25:
Thanks, I will try.

closed 2018-08-25 by hawaiianer