ArduPilot/MissionPlanner #1929: MAVlink inspector can‘t measure packet rate more than 16,7 Hz

suraqis, 2018-08-22:
#### Issue details
MAVlink inspector available from ctrl+f menu can’t measure packet rates higher than 16,7 Hz for certain type of packet. Actually streaming rates are limited to 10 Hz (except ADS-B), but Ardupilot allow to change rate higher than 10, up to 50 Hz, but didn’t save the rate. So when the rate is higher than 16,7 it shows only 16,7. Double checked with oscilloscope. 

#### Version
MP 1.3.58

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
Any

#### Logs
Reproducable

meee1, 2018-08-22:
ok found the issue. I only allow a max of 50 packets in a 3 seconds period. ill change to 175

meee1, 2018-08-22:
fixed in beta

closed 2018-08-23 by meee1