ArduPilot/MissionPlanner #1931: Make MP override SRN_XXX rates optional

suraqis, 2018-08-23:
#### Issue details
Mission Planner overrides SRN_XXX data rates on AP's serial interface through which it's connected with values set in "Planner" tab of "Config/tuning" view. It makes setting up desired data rates on certain interfaces of autopilot or other mavlink devices (custom ones) confusing. Overriding AP parameters for SRN_XXX in MP settings should be optional.

#### Version
MP 1.3.58

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
All

#### Hardware type
Any

#### Logs
No

meee1, 2018-08-23:
MP only changes the interface its connected to. so I would suggest setting the options in planner settings

suraqis, 2018-08-23:
If I connect MP to system with multiple mavlink devices and change device under control with sys/comp dropdown menu, will MP rate settings affect remote mavlink device SR parameter? 

When I set up the whole system I need to connect to each device (to setup baudrate, sys/comp id, etc.) before I interconnect them into one system. Do I need to change SR rates on each device after I once setup it with direct connection?

Anyway, overriding SR rates from MP seems not obvious for me and should be optional with check box i MP configuration.

meee1, 2018-08-23:
mp will set the current active mav only. ie using the drop down to change which AP you are talking to will set those as well. 
what is your desired state? 
MP job is to report flight status, so therefor it ensures the selected rates are archived. if its not getting data, but the rates are set to, it will request that data. at the same time, if you set mp to 0, it will request the data at a rate of 0, and then because that target will be archived, wont request it again.