ArduPilot/MissionPlanner #1938: CS Values Accessed in Python Script Not being Updated

Quigleyrw, 2018-09-05:
#### Issue details
If a python script is started before connecting an aircraft (as occurs when conducting a command line launch with the -script flag) or starting a simulation some cs values (cs.lat, cs.lng, etc.) do not properly update. If no craft or simulation has been connected in this run of Mission Planner the reported values for latitude and longitude are both 0. If a user connects an aircraft, runs a script, disconnects the aircraft and then reconnects values are repeated. These appear to be the last values prior to the connection interruption.  If you stop the script and re-run it after a flight controller is connected the issue resolves itself. 

This issue was observed using both a quadcopter simulation, and pixhawk 2 flight controller

Sample Code For Testing:
`while cs.connected is False:
    print "Waiting for Platform"
    Script.Sleep(1000)

while True:
    print "LAT:", cs.lat
    print "LNG:", cs.lng
    Script.Sleep(1000)`

#### Version
1.3.53 and 1.3.58

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

Note: Not tested with other platforms

#### Airframe type
quad simulation

#### Hardware type
pixhawk2

meee1, 2018-09-06:
are you starting your script before you connect a drone? or after.

you will have different results depending on your answer

Quigleyrw, 2018-09-06:
Before connecting a drone.

The basic idea was to use the -script flag when launching Mission Planner to prevent having to go to the effort of finding, loading, and running a chosen script every time.

meee1, 2018-09-06:
ok, in that case I would not use cs
use either Ports or MAV
ie
MAV.cs
Ports[0].MAV.cs

meee1, 2018-09-06:
this is all so you can handle multiple drones at any one time

Quigleyrw, 2018-09-06:
No, the script's purpose in this case is just collection of data from a platform during a flight. 

Is there any documentation on using Ports or MAV as you suggested for this purpose that you might recommend looking at?

meee1, 2018-09-14:
there are some exampled in the scripts directory in the MP folder. other than that you can use dir(variable) to discover new features and refer to the MP code