ArduPilot/MissionPlanner #1941: Error when close vlc stream

MonkO, 2018-09-10:
Hello!
After vlc rtsp stream,i close MP, and have error:
[image]

How can i fix it?

meee1, 2018-09-14:
ive added a bitmap clone, so should prevent this error, but will reduce performance.

closed 2018-09-14 by meee1

MonkO, 2018-09-24:
I try to use bitmap clone, but it didnt help me. Can you show me where, you mean, use it?

also im interested to use gstreamer, but pipeline what i use didnt work:
rtspsrc location=rtsp://192.168.10.145:554/mux1.sdp ! queue ! rtpjitterbuffer latency=100 ! rtph264depay ! avdec_h264 ! autovideosink sync=false
but it work through cmd