ArduPilot/MissionPlanner #1943: 1.3.58 always crashes at startup with error shown under mono on Ubuntu 16.04

laurenweinstein1, 2018-09-13:
[image]

meee1, 2018-09-13:
update your mono version

laurenweinstein1, 2018-09-13:
Thanks. That got me much further. I thought that machine's mono was up to date but somehow it slipped. With the mono update, MP opens, and immediately opens a UDP connection to a SkyViper Journey (I never could get any previous MP versions to connect at all even with the Journey set to MAVLINK version 1). MP also detected the video stream, but fails (probably as expected) at the automatic gstreamer install. There is another problem though. MP pulls in all of the parameters from the Journey, and pops up a Connecting Mavlink "Done" box. But then the GUI freezes. The HUD is populated with (static) data, the Done box can be moved around. but the interface itself is completely frozen.

laurenweinstein1, 2018-09-14:
In fact, on further inspection, I see the HUD actually keeps updating (even though the lower panel with the values is grayed out -- the values are changing -- for several seconds at least after the Done box pops up. Then it all freezes up completely and has to be killed externally.

laurenweinstein1, 2018-09-14:
I have it working now. The sequencing of MP startup and the UDP source are touchy, but I have a workable solution. Thanks!

closed 2018-09-14 by laurenweinstein1

meee1, 2018-09-14:
@laurenweinstein1 good to hear that!!

laurenweinstein1, 2018-09-14:
It's rather fragile. If the Wi-Fi connection is established before MP starts, it will freeze up every time at the Done pop-up. If I let MP start and then once it's going bring up the connection, it will usually work, sometimes immediately pulling in all parameters and completing, or sometimes requiring that I hit Connect manually. Sometimes all of the parameters don't come in, but ultimately after a couple of tries they do. When doing it this way, the gstreamer setup message never appears.

meee1, 2018-09-15:
can on get some details on your specific setup
ie I gather its udp client?
the autoconnect thing is on a MP startup timer, ie first 30 seconds since MP started, after that its manual.

give when its sending and it fails, this all could be related to the first 30 seconds.

laurenweinstein1, 2018-09-15:
The ArduCopter side is acting as AP, so the MP side is the UDP client. I agree, it's the startup sequence timing in those first 30 seconds or so that seem to be crucial.