ArduPilot/MissionPlanner #1944: Selectable Focal lenght in Survey(Grid) max 180mm

imarqs, 2018-09-14:
#### Issue details
To plan mapping missions the camera focal length has to be selected in Mission Planer's Survey Plugin. The maximum selectable Focal length is 180mm. At least the Hasselblad A6D can be equipped with lenses up to 300mm.

#### Version
Mission Planner 1.3.58

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Mission Planner

#### Hardware type
Windows Ground Station

#### Logs
-

meee1, 2018-09-14:
just because I have to ask, how high are you flying/whats your target ground resolution if you need to use 300mm?
most people stick in the 16-35mm range for doing grid surveys. having a 300mm zoom would need a very high capture rate to get any cross coverage.

imarqs, 2018-09-14:
Yes you are right! 
I plan to achieve 1mm resolution and can't fly as low as I want, that's why tele lenses are the only way to go. I should add that I try to map railway tracks before I sound to creepy ;)

And due to motion blurring I can't fly more than 1m/s anyway, so capture rate is not the biggest problem. 

Mission Planner helped me a lot figuring all that out btw!

meee1, 2018-09-14:
ive made a change in beta for up to 500mm

closed 2018-09-14 by meee1