ArduPilot/MissionPlanner #1949: Pixhawk4 cannot load firmware on Mission Planner

Fishton, 2018-09-27:


meee1, 2018-09-27:
please use the beta firmware option

Fishton, 2018-09-27:
Does the same thing Mike
[image]

meee1, 2018-09-27:
what MP version?

Fishton, 2018-09-27:
1.3.58.1

meee1, 2018-09-27:
what firmware are you trying to install? copter/plane etc

Fishton, 2018-09-27:
Rover

Fishton, 2018-09-27:
https://www.facebook.com/john.easton3/videos/10156759294201202/

meee1, 2018-09-27:
it appears that there has been no stable or beta release's that support that board.
so you need to use master.

goto firmware screen. press control-q, then try rover

Fishton, 2018-09-27:
When do you press Ctrl - q?

meee1, 2018-09-27:
at the firmware screen

Fishton, 2018-09-27:
with the board connected via usb obviously?

Fishton, 2018-09-27:
Nothing happens when I push Ctrl-q, it just does the same thing

Fishton, 2018-09-27:
When I plug my Pixhawk2.1 in it works perfectly

meee1, 2018-09-27:
try updating to beta MP

help > update beta

Fishton, 2018-09-27:
trying now ...

Fishton, 2018-09-27:
updating ...

Fishton, 2018-09-27:
It says - "Update Failed - Operation timed out"

Fishton, 2018-09-27:
No luck

Fishton, 2018-09-27:
Confirm - Pixhawk4 does not work.

Fishton, 2018-09-27:
What is supposed to happen when I press Ctrl-q?

Fishton, 2018-09-27:
I have tested mine and my supplier have tested theirs and we are experiencing the same problem, so please explain this CTRL-q thing please Michael. Thank you

meee1, 2018-09-27:
open the install firmware screen, press control-q. if it doesn't do anything, click anywhere on the screen and press control-q again.

Fishton, 2018-09-27:
Uh-Huh!! That did it!  Thank you.  I would never have thought of that.

Fishton, 2018-09-27:
ERROR: Firmware not suitable for this board

Fishton, 2018-09-27:
This was for Rover Dev