ArduPilot/MissionPlanner #1957: Simulator unable to load terrain data

rmackay9, 2018-10-11:
while testing at a search and rescue event in Hokkaido Japan I found that the simulator was unable to load terrain data.  This means we are unable to test the terrain alt based missions.
[image]

mission: https://www.dropbox.com/s/wuxrd9uttva1nsf/JIC-test-mission.waypoints?dl=0

meee1, 2018-10-11:
I have seen this as well, but given mission planner works for real flights, I don't think its a MP issue.
its a sitl issue I think.

meee1, 2018-11-16:
I think I found the source of this issue.
the mavlink packet spec changed, and MP didn't handle the change to the packet structure

closed 2018-11-22 by meee1