ArduPilot/MissionPlanner #1958: Gstreamer RTSP stream pipeline problem

MonkO, 2018-10-12:
Hi!
I'm trying receive my RTSP video on MissionPlanner. I use this pipeline:
rtspsrc location=rtsp://192.168.10.145:554/mux1.sdp ! application/x-rtp ! rtph264depay ! avdec_h264 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink
it work, when I put it on cmd, but not in MissionPlanner.
Also I trying  udp like in example:
videotestsrc pattern=ball ! x264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5600
and it work. But I need rtsp video. Can you help me?

meee1, 2018-10-20:
what error do you get when you do it in MP?

meee1, 2018-10-22:
to use
rtspsrc location=rtsp://192.168.10.145:554/mux1.sdp ! application/x-rtp ! rtph264depay ! avdec_h264 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink

you need to right click the hud > set gstreamer source. and paste in that pipeline

MonkO, 2018-10-23:
I right click the hud and set gstreamer source. but it not working. No error, but no video too

meee1, 2018-10-25:
please try the latest version. i found a bug

closed 2018-11-05 by meee1