ArduPilot/MissionPlanner #1967: Issues with building Mission Planner with VS 2017

peilibuaa, 2018-10-23:
#### Issue details
I got a lot of errors about "Meta data *.dll could not be found" when building the solution with VS2017.

Here is the ERROR LIST

[image]
[image]

#### Version
I build the Mission Planner git clone from "https://github.com/ArduPilot/MissionPlanner"

Window 10 home 1803

Visual Studio Community 2017 15.8.7

Anyone could give me some hints on how to remove the build errors? Thank you very much.

ZwiiDoan, 2018-11-02:
I don't have that many errors but also got this:
`Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3073	The command "dotnet "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.build\0.6.0\targets\../tools/illink/illink.dll" -l none --verbose --strip-security true --exclude-feature com --exclude-feature sre -v false -c link -u link -b true -d "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.build\0.6.0\targets\../tools/mono/bcl/" -d "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.build\0.6.0\targets\../tools/mono/bcl/Facades/" -o "C:\Users\DuyDoan\Documents\GitHub\MissionPlanner\ExtLibs\wasm\obj\Debug\netstandard2.0\blazor\linker/" -x "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.build\0.6.0\targets\BuiltInBclLinkerDescriptor.xml" -x "C:\Users\DuyDoan\Documents\GitHub\MissionPlanner\ExtLibs\wasm\obj\Debug\netstandard2.0\blazor\linker.descriptor.xml" -a "C:\Users\DuyDoan\.nuget\packages\bitmiracle.libtiff.net\2.4.584.1\lib\netstandard1.3\BitMiracle.LibTiff.NetStandard.dll" -a "C:\Users\DuyDoan\.nuget\packages\blazor.extensions.canvas\0.1.7\lib\netstandard2.0\Blazor.Extensions.Canvas.JS.dll" -a "C:\Users\DuyDoan\.nuget\packages\blazor.extensions.canvas\0.1.7\lib\netstandard2.0\Blazor.Extensions.Canvas.dll" -a "C:\Users\DuyDoan\.nuget\packages\blazor.extensions.signalr\0.1.7\lib\netstandard2.0\Blazor.Extensions.SignalR.JS.dll" -a "C:\Users\DuyDoan\.nuget\packages\blazor.extensions.signalr\0.1.7\lib\netstandard2.0\Blazor.Extensions.SignalR.dll" -a "C:\Users\DuyDoan\.nuget\packages\csmatio\1.0.20\lib\netstandard2.0\csmatio.dll" -a "C:\Users\DuyDoan\.nuget\packages\flurl\2.8.0\lib\netstandard2.0\Flurl.dll" -a "C:\Users\DuyDoan\.nuget\packages\flurl.http\2.3.2\lib\netstandard2.0\Flurl.Http.dll" -a "C:\Users\DuyDoan\.nuget\packages\geoapi.coordinatesystems\1.7.5\lib\netstandard2.0\GeoAPI.CoordinateSystems.dll" -a "C:\Users\DuyDoan\.nuget\packages\geoapi.core\1.7.5\lib\netstandard2.0\GeoAPI.dll" -a "C:\Users\DuyDoan\.nuget\packages\log4net\2.0.8\lib\netstandard1.3\log4net.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor\0.6.0\lib\netstandard2.0\Microsoft.AspNetCore.Blazor.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.browser\0.6.0\lib\netstandard2.0\Microsoft.AspNetCore.Blazor.Browser.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.build\0.6.0\lib\netstandard1.0\Microsoft.AspNetCore.Blazor.TagHelperWorkaround.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.csharp\4.5.0\lib\netstandard2.0\Microsoft.CSharp.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.extensions.dependencyinjection\2.1.0\lib\netstandard2.0\Microsoft.Extensions.DependencyInjection.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.extensions.dependencyinjection.abstractions\2.1.0\lib\netstandard2.0\Microsoft.Extensions.DependencyInjection.Abstractions.dll" -a "C:\Users\DuyDoan\.nuget\packages\microsoft.jsinterop\0.6.0\lib\netstandard2.0\Microsoft.JSInterop.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\microsoft.win32.registry\4.5.0\lib\netstandard2.0\Microsoft.Win32.Registry.dll" -a "C:\Users\DuyDoan\.nuget\packages\mono.webassembly.interop\0.6.0\lib\netstandard2.0\Mono.WebAssembly.Interop.dll" -a "C:\Users\DuyDoan\.nuget\packages\newtonsoft.json\11.0.2\lib\netstandard2.0\Newtonsoft.Json.dll" -a "C:\Users\DuyDoan\.nuget\packages\opentk\3.0.1\lib\net20\OpenTK.dll" -a "C:\Users\DuyDoan\.nuget\packages\ssh.net\2016.1.0\lib\netstandard1.3\Renci.SshNet.dll" -a "C:\Users\DuyDoan\.nuget\packages\sshnet.security.cryptography\1.2.0\lib\netstandard1.3\SshNet.Security.Cryptography.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.appcontext\4.1.0\lib\netstandard1.6\System.AppContext.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.buffers\4.4.0\lib\netstandard2.0\System.Buffers.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.collections.concurrent\4.3.0\lib\netstandard1.3\System.Collections.Concurrent.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.collections.immutable\1.2.0\lib\netstandard1.0\System.Collections.Immutable.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.collections.nongeneric\4.0.1\lib\netstandard1.3\System.Collections.NonGeneric.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.diagnostics.diagnosticsource\4.3.0\lib\netstandard1.3\System.Diagnostics.DiagnosticSource.dll" -a "C:\Users\DuyDoan\.nuget\packages\system.diagnostics.stacktrace\4.0.1\lib\netstandard1.3\System.Diagnostics.StackTrace.dll" -a "C:\Users\DuyDoan\.nuget\packages\system.drawing.common\4.5.0\lib\netstandard2.0\System.Drawing.Common.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.io.filesystem.primitives\4.3.0\lib\netstandard1.3\System.IO.FileSystem.Primitives.dll" -a "C:\Users\DuyDoan\.nuget\packages\system.io.ports\4.5.0\lib\netstandard2.0\System.IO.Ports.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.linq\4.3.0\lib\netstandard1.6\System.Linq.dll" -a "C:\Users\DuyDoan\.nuget\packages\system.memory\4.5.0\lib\netstandard2.0\System.Memory.dll" -a "C:\Users\DuyDoan\.nuget\packages\system.net.webheadercollection\4.0.1\lib\netstandard1.3\System.Net.WebHeaderCollection.dll" -a "C:\Program Files\dotnet\sdk\NuGetFallbackFolder\system.numerics.vectors\4.4.0\lib\netstandard2.0\System.Numerics.Vectors.dll" -a "C:\Users\DuyDoan\.nuget\packages\system.reactive\4.1.0\lib\netstandard2.0\System.Reactiv [... cut ...] an\Documents\GitHub\MissionPlanner\ExtLibs\Strings\bin\Debug\netstandard2.0\zh-Hans\MissionPlanner.Strings.resources.dll" -a "C:\Users\DuyDoan\Documents\GitHub\MissionPlanner\ExtLibs\GMap.NET.Core\bin\Debug\netstandard2.0\zh-Hans\GMap.NET.Core.resources.dll" -a "C:\Users\DuyDoan\Documents\GitHub\MissionPlanner\ExtLibs\wasm\obj\Debug\netstandard2.0\wasm.dll"" exited with code 1.	wasm	C:\Users\DuyDoan\.nuget\packages\microsoft.aspnetcore.blazor.build\0.6.0\targets\Blazor.MonoRuntime.targets	446	
`

RisingPhoenix247, 2018-11-09:
Download the missing dll files from the master and paste each of  them in the respective folders.