ArduPilot/MissionPlanner #1972: Joystick sending stale data as RC_OVERRIDE?

rmackay9, 2018-10-31:
I've been looking into an issue reported by @jmachuca77 and NickN regarding their rovers jumping forward for a moment after **recovering** from a radio or GCS failsafe.

After a fair bit of investigation on the AP Rover side I think the issue is actually in the MP.  The way to reproduce the issue is:

1. load Rover-3.5.0-dev firmware to a rover
2. set MOT_THR_SLEWRATE = 0 (to remove any throttle output slewing which makes the problem more obvious)
3. connect with MP and attach a joystick, ensure the joystick is working by going to the InitialSetup >> Mandatory Hardware >> Radio Calibration page
4. switch the vehicle to Manual mode, arm
5. increase the joystick's throttle to 100% and keep it there
6. disconnect the joystick from the computer.  the MP will display a "lost joystick" window.  After a few seconds the Rover will likely enter failsafe, switch to Hold mode and stop the motors
7. switch the vehicle back to Manual mode (even though it's in failsafe)
8. reduce joystick's throttle back to middle, reconnect the joystick
9. on MP's joystick setup page, push the "Disable" button (it would make more sense if this button was switched to "Enable" when it lost contact with the joystick but that's a separate issue).  At this moment the RC input received by AP jumps to the throttle value before the joystick was unplugged.  This is sent to the vehicle and the vehicle's motors will momentarily pulse forward.

I've put a [tlog of my tests here](https://www.dropbox.com/s/o63rghwc02v2jod/rover-rcoverride-glitch.tlog?dl=0).

meee1, 2018-11-01:
this should be fixed in beta

meee1, 2018-11-01:
update here https://github.com/ArduPilot/MissionPlanner/commit/38b32090d087166d845df065e2a29dd7a6300964

rmackay9, 2018-11-02:
@meee1, great, thanks!!

rmackay9, 2018-12-24:
I think this is fixed now so closing, txs!

closed 2018-12-24 by rmackay9