ArduPilot/MissionPlanner #1973: optimize GUI for parameter changes (in Config Tuning > Full parameter list)

aercamti, 2018-11-01:
#### Issue details
A couple suggestions for improvements for working with parameter changes in Mission Planner:

1) When making changes on parameters, the lists get's really long and difficult to handle. Please consider adding either 
- a checkbox to show (filter) only those items that have been changed, but not written to the FC yet (the green ones)
- or a "sort by" column
This would greatly help in getting a quick overview on only the changed items and thereby reduce potential errors.

2) When importing parameters from a file, a warning prompt will come for some items: "... is marked as read only and should not be changed"
Please consider adding an option to select _which_  parameters not to overwrite. 
Or if that is too complicated/ cumbersome, simply add a box asking: 
- leave all read-only items at their current values
- overwrite all read-only items with the imported values

3) When importing a *.param file (using the full parameter list section), there is no confirmation for successful write. (The other views do show a confirmation box).
Please consider adding a confirmation in this view as well. This would avoid ambiguity.

Thanks!
#### Version
1.3.59.2
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Hexacopter
#### Hardware type
Pixhawk 2.1
#### Logs
n/a

aercamti, 2018-11-06:
I am familiar with the compare option, which already are a considerable help.
The above suggestions are meant to make working with the software more user-friendly and to reduce the possibility of (over-)writing wrong entries.

Naterater, 2018-12-02:
I agree that "green" parameters that are about to be written should be able to be sorted in some way. I think moving them to the top or bottom of the screen and staying there until they are successfully written would avoid ambiguity and increase user-friendliness and confidence with working with full parameter lists. Because of the ever-changing nature of ardupilot, the full parameter list is going to be used for a long time. It's worth it to make these sort of changes.

meee1, 2018-12-02:
1. ive added a modified checkbox. 
2. ive made all readonly params not imported. as they are read only. if this is not the case its an underlying issue that should be address elsewhere.
3. im confused, as im not sure what im actually confirming.

Naterater, 2018-12-02:
Thank you for addressing these, especially #2.
[image]

For #3, a textbox confirmation that the parameter write was successful would be nice. Sometimes parameter transfer takes a while, and the only way to tell if it's done is by watching for the "Write Params" button turn from light green to dark green.

aercamti, 2018-12-02:
Yes, thank you for implementing those changes!
With #3 I meant to please add a confirmation box popping up following a successful write.
When writing params in the "Standard Params" view, there is a such a confirmation box, headed with "Saved" and the text "Parameters successfully saved".
[image]

This would also make sense for the "Full Parameter List" view, as Naterater elaborated.

aercamti, 2018-12-12:
Thanks for optimizing these three points so quickly!

Topping this off (for other users) would be a **tool-tip** which explains the function of the "Modified" check box, such as:
_"Check this box to display only parameters which have been modified; uncheck to show all params again."_