ArduPilot/MissionPlanner #1982: Add flag to let user set directory for caches

wucke13, 2018-11-07:
#### Issue details
I'm currently working on a linux compatible `nixpkg` for Mission Planner. The package is almost fine, though Mission Planner crashes on startup complaining not having the permission to create the `gmapcache` folder. It is by design not allowed to change content of a program directory after the program was installed for `nixpkgs`, and I would guess it is considered bad practice too to mix static data (executables, libraries, etc.) with dynamic data (user data, caches, etc.).

So, please give us a way to specify a prefix/folderpath for all dynamic data (files and folders that are going to change), preferable through an environment variable or a CLI flag.

#### Version 
1.3.59

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2018-11-07:
whats the full path to the gmap folder its trying to use?

on windows this folder is separate, so the issue is more likely where mono allocated the folder.

meee1, 2018-11-07:
ive just made a change in beta to use a directory in the users home folder

wucke13, 2018-11-07:
Mission Planner tries to create `gmapcache` folder directly in the folder containing `MissionPlanner.exe`

meee1, 2018-11-08:
you can grab the beta from here
https://ci.appveyor.com/project/meee1/missionplanner/builds/20132662/artifacts

meee1, 2018-11-08:
one possible performance tweak as well would be to do a 

mono --aot MissionPlanner.exe prior to running
there is a aot.sh script that does this. might have the wrong line endings though

closed 2018-12-25 by wucke13