ArduPilot/MissionPlanner #1985: TCP Mavlink Mirror/pass trough not working on current version

pompecukor, 2018-11-09:
#### Issue details
TCP Mavlink Mirror/pass trough not working on current version.
Used to work on previous versions (was 6 months ago since I last tried, so not sure which version he bug was introduced.

It is very easy to reproduce. Start a Tlog or even just a simulation. Open Advance options (Cont+F). click on Mavlink. Select TCP host. Select baud 57600 (in our case). Click start.
No stream!

#### Version
Current

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

tridge, 2018-11-09:
I've tested this against mavproxy and it fails for me too

meee1, 2018-11-09:
worked first time for me?
did you tick write access or not?

pompecukor, 2018-11-09:
> did you tick write access or not?

That did the trick. Don't remember needing that before. Thanks for the fast response.
May I recommend having it ticked by default and option to remove.
I am not sure which is more common use case.
Anyway I am very happy. Thx

pompecukor, 2018-11-09:
Solved: Need to tick write access!

closed 2018-11-09 by pompecukor

tridge, 2018-11-09:
still doesn't work for me - maybe I just don't know how to use it?

I did this:

- started with MissionPlanner disconnected
- control-F
- choose Mavlink button
- choose "TCP Host - 14550"
- choose random baudrate
- enable write access tickbox
- click connect
- in mavproxy do "output add tcp:192.168.2.48:14550" (IP is my windows box)

that leaves me with an establshed tcp connection, so clearly MP is listening and accepting the TCP connection. What I don't get is MP connecting, asking for parameters etc. I don't see in the UI how to ask MP to actually use this new TCP connection it has established.
Any ideas?

meee1, 2018-11-09:
@tridge the idea of the mavlink button is to transport traffic to a 3rd party. not receive from a third party. 
ie the drone is connected to MP as normal. and then you want a 2nd instance of MP, and you connected it using this method.
if you want to receive from a 3rd party, then you would just connect as normal.

tridge, 2018-11-09:
ahh, ok, I did misunderstand it then, thanks Michael

Asadsaeed003091, 2022-11-15:
> worked first time for me? did you tick write access or not?

TCP Mavlink Mirror is working for me by selecting TCP HOST - 14550, As a curoristy i am trying to do the same with UDP - 14450. It not working with UDP. It ask me to enter local port (ensure remote end is already sending). I am confused about it as i have to establish stream

nikhil1249, 2024-07-19:
Anyone able to solve this issue reported by [Asadsaeed003091](https://github.com/Asadsaeed003091)