ArduPilot/MissionPlanner #1990: MP 1.3.59 EKF errors with Omnibus F4 Pro V2

RANNDON, 2018-11-16:
#### Issue details
EKF horizontal velocity and position errors when using MP 1.3.59 - OK with MP 1.3.58. The fault prevents Rudder Arming.
#### Version
1.3.59
#### Platform
[X ] Plane
#### Airframe type
EP Monoplane
#### Hardware type
Omnibus F4 Pro V2
#### Logs
Screenshot PDF and Dataflash log renamed as a TXT file attached:

[image]

[00000002.txt](https://github.com/ArduPilot/MissionPlanner/files/2587836/00000002.txt)

meee1, 2018-11-16:
its not an error. the ekf has not finish initializing yet.

RANNDON, 2018-11-16:
Michael,

 

I admire you work and I appreciate your rapid response.

 

The performance between MP 1.3.58 and 1.3.59 is significantly different. I have tested it and so has my friend in Bundaberg. We both see the same results.

 

It may not be an MP issue, but the symptoms are clear. I thought it was important that I flag the issue.

 

Regards, Don

meee1, 2018-11-16:
if you are arming from the rc transmitter. then MP only shows you what the autopilot is doing. Nothing is sent from MP to the drone. ie nothing to do with missionplanner other than displaying what the autopilot is telling it.

RANNDON, 2018-11-16:
> what is your workaround? as I don't believe it should make any difference based on the version of MP

Michael,

 

All I know is that I get no EKF errors on 1.3.58 and EKF errors on 1.3.59.

 

A friend of mine in Bundaberg liked my Omnibus/Arduplane combo, so I tried to walk him through the build remotely, from my base in Canberra. All went well until we tried to Rudder-Arm his FC. We discovered that EKF errors were preventing Arming.

 

My Omnibus was working, so he shipped his down to Canberra so I could test it. To cut a long story short, we discovered that MP 1.3.59 was the difference. My mate had the same problem on two separate FCs. I was running 1.3.58 and he was running 1.3.59. He downgraded to 1.3.58 and it worked.

 

I received my mates FC and tested it ok on 1.3.58. I upgraded to 1.4.59 and hey-presto the problem re-appeared. Back to 1.3.58 and the problem is gone.

 

I can only report upon what I see. I have three more Omnibus FCs arriving early next week. I will report their performance if that us helpful.

 

Regards, Don

meee1, 2018-11-16:
the error is cause by a bad gps position. are you testing with good gps reception?

RANNDON, 2018-11-16:
Yes

RANNDON, 2018-11-16:
Michael,

 

Are you in Australia? I will express-post you a failing FC if you are.

 

Regards, Don

meee1, 2018-11-16:
yes I am, but the red EKF status you talk about is a display change I made between .58 and .59 
the problem still existed in .58, it just didn't show it like it does I .59
im just trying to determine if it is anything at all to do with the version, but I don't believe it is.

RANNDON, 2018-11-16:
Michael,

 

I will contact you again when I have a failing FC ready to ship. You can let me know then whether you want me to ship.

 

Thanks for your follow-up.

 

Regards, Don

RANNDON, 2018-11-20:
Michael, I have another piece of the puzzle for you. I have been using my Omnibus F4 Pro FCs with compass disabled. I decided to test the effect of adding a compass. The presence or absence of a FC compass makes no difference to the EKF status under MP 1.3.58. However, under MP 1.3.59 the absence of a compass causes solid EKF horizontal velocity and position errors. If I add a compass under 1.3.59, I get EKF errors for about 20 seconds after I boot up, then the errors clear. So, it is a matter of whether I stay with MP 1.3.58 and no compass, add a compass and move on, or disable EKF and try my hand with DCM with no compass.

meee1, 2018-11-20:
yes, that makes sence.
the EKF even though it doesn't display red in 1.3.58 is still unhappy. where as 1.3.59 shows that it is indeed unhappy.

RANNDON, 2018-11-20:
Thanks Michael. I will stay with MP 1.3.58 and no compass. The plane is flying great. I will test DCM when and if I need to move on.