ArduPilot/MissionPlanner #2000: Tuning: add flow_x, flow_y for raw values from optical flow sensor

rmackay9, 2018-11-24:
We are planning to add support for this optical flow sensor which will probably increase the number of users of the flow sensors significantly (assuming they work well).

It would be helpful if the sensor's raw x and y values ([OPTICAL_FLOW.flow_x and flow_y](https://mavlink.io/en/messages/common.html#OPTICAL_FLOW)) were easily visible to the user in the status and/or tuning.  At the moment only the compensated values are visible which makes it harder to see the sensor is working because this includes the gyro data (in fact there appears to be a bug in our reporting so currently we are only sending gyro data in these fields which is not useful)

meee1, 2018-11-29:
they are already there, under opt_x and opt_y

meee1, 2018-11-29:
the messages I pull out currently are
               var optflow = mavLinkMessage.ToStructure<MAVLink.mavlink_optical_flow_t>();
// left is status screen > right is mavlink message field
                        opt_m_x = optflow.flow_comp_m_x;
                        opt_m_y = optflow.flow_comp_m_y;
                        opt_x = optflow.flow_x;
                        opt_y = optflow.flow_y;
                        opt_qua = optflow.quality;

rmackay9, 2018-11-30:
Maybe there is a difference between the status page (which I'm not using) and the Tuning page (which I'm using)?
[image]

meee1, 2018-11-30:
agh ok, looks like i might have to check why its not appearing in that list

[mp-tuning-page]<https://user-images.githubusercontent.com/1498098/49260154-9d301c80-f47f-11e8-8a31-1f3c774eff26.png>

meee1, 2018-12-01:
this is now in beta

closed 2018-12-01 by meee1

rmackay9, 2018-12-01:
Great, thanks!!