ArduPilot/MissionPlanner #2004: Mission planner dont launch

Kirilpython, 2018-11-26:
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
Data Dir C:\ProgramData\Mission Planner\
Log Dir C:\Users\UAV-LAB\Documents\Mission Planner\logs
Running Dir C:\Users\UAV-LAB\Desktop\MissionPlanner-master\bin\Debug\net461\
User Data Dir C:\Users\UAV-LAB\Documents\Mission Planner\
INFO Program - ******************* Logging Configured *******************
INFO Program - 64bit os True, 64bit process True
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883_OLD or 1
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883 or 7
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype 17 or DEVTYPE_ACC_LSM303D
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 36 or DEVTYPE_GYR_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 4 devtype 34 or DEVTYPE_GYR_L3GD20
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 36 or DEVTYPE_GYR_MPU9250
FATAL Program - Fatal app exception
System.TypeInitializationException: The type initializer for 'MissionPlanner.MainV2' threw an exception. ---> System.IO.FileLoadException: Could not load file or assembly 'SvgNet, Version=1.0.0.0, Culture=neutral, PublicKeyToken=38326cb7e06851fc' or one of its dependencies. Strong name validation failed. (Exception from HRESULT: 0x8013141A) ---> System.Security.SecurityException: Strong name validation failed. (Exception from HRESULT: 0x8013141A)
   --- End of inner exception stack trace ---
   at MissionPlanner.MAVState..ctor(MAVLinkInterface mavLinkInterface, Byte sysid, Byte compid)
   at MissionPlanner.Mavlink.MAVList..ctor(MAVLinkInterface mavLinkInterface)
   at MissionPlanner.MAVLinkInterface..ctor()
   at MissionPlanner.MainV2..cctor()
   --- End of inner exception stack trace ---
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Main(String[] args)
System.TypeInitializationException: The type initializer for 'MissionPlanner.MainV2' threw an exception. ---> System.IO.FileLoadException: Could not load file or assembly 'SvgNet, Version=1.0.0.0, Culture=neutral, PublicKeyToken=38326cb7e06851fc' or one of its dependencies. Strong name validation failed. (Exception from HRESULT: 0x8013141A) ---> System.Security.SecurityException: Strong name validation failed. (Exception from HRESULT: 0x8013141A)
   --- End of inner exception stack trace ---
   at MissionPlanner.MAVState..ctor(MAVLinkInterface mavLinkInterface, Byte sysid, Byte compid)
   at MissionPlanner.Mavlink.MAVList..ctor(MAVLinkInterface mavLinkInterface)
   at MissionPlanner.MAVLinkInterface..ctor()
   at MissionPlanner.MainV2..cctor()
   --- End of inner exception stack trace ---
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Main(String[] args)

Press any key to exit!

_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2018-11-29:
please update from the msi

closed 2018-11-29 by meee1

uavmaster, 2018-11-29:
Hello