ArduPilot/MissionPlanner #2006: Mission Planner 1.3.6x does not open Logs

Franky464, 2018-11-27:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2018-11-27:
are you using .60 or .61?

vierfuffzig, 2018-11-27:
mine first did refuse to display logs too after updating to 3.61 build 1.3.6905.12345. after having having updated to latest beta 1.3.61 build 1.3.6904.21948 logs are back, but in-graph display of mode, errors and MSG is gone.
[image]

as opposed to .60:
[image]

basti

Franky464, 2018-11-28:
Well MP build 1.3.6904.12435 opens a black window all values @ zéro, obviously it cannot open the file.
either with .log or .bin files
MP Build 1.3.6877.26630 does it fine, i beleive a bug appeared on lastupdate and previous.
 
[image]

vierfuffzig, 2018-11-28:
@franky464 your screenshot looks like the log file is actually loaded as you have the value column on the right produced correctly. 
beyond that the data table checkbox has changed to deactivated by default now. that may add some confusion too.

Franky464, 2018-11-28:
Vierfuffzig, 
Thanks for the feed back, i'm feeling stupid ;o( this is exactly the thing the checkbox  was not activated, the message are back with the beta 1.3.6905.43023.