ArduPilot/MissionPlanner #2007: Dataflash log Auto Analysis tool is out of date for Copter-3.6

rmackay9, 2018-11-28:
Users are reporting a number of issues with the Auto Analysis tool when used with logs from Copter-3.6.x ([discussion is here](https://discuss.ardupilot.org/t/problems-after-flashing-v3-6-1/35550)):

- The value of LOG_BITMASK is 655358, which is Full + IMU. But log analysis says “No IMU log data”.
- "Found NaN in CTUN DSAlt" even though this is expected when there is no sonar attached
- “THR_MIN not found" because this has been renamed to MOT_SPIN_MIN

meee1, 2018-11-28:
the log analysis is just a copy of this
https://github.com/ArduPilot/ardupilot/tree/master/Tools/LogAnalyzer

so would need updating as a whole.

rmackay9, 2018-11-28:
yes, I agree it's not really an MP issue.  I'll re-raise this over in the ardupilot area.

closed 2018-11-28 by rmackay9