ArduPilot/MissionPlanner #2009: Telemetry rates being reset to value when first connected?

rmackay9, 2018-11-29:
I'm seeing something a bit odd with the "Telemetry Rates" and the actual update rate from the vehicle.

To recreate the issue do this:

1. connect to a flight controller running Rover-3.5.0-dev (other versions probably also work)
2. set the Config Tuning >> Planner >> Telemetry Rates to 10 (or anything different that what you see at this point)
3. switch to the Flight Data screen and move the flight controller around to convince yourself that the update rate is similar to what you just selected above
4. disconnect and reconnect and switch back to the Planner screen and check the Telemetry Rates

In my tests, at Step 4 the rates have returned to what they were before I changed them in Step 2.  It's possible that this is the desired behaviour but it seems odd because the flight controller hasn't been rebooted so it seems like the MP has somehow remember the original Telemetry Rates and send a command to the vehicle to set the rates.

meee1, 2018-11-29:
there was a reason for this. I don't recall why now though.

meee1, 2018-12-01:
@Naterater MP does not request any changes if you have set the options to -1.

Naterater, 2018-12-01:
Is that documented or noted anywhere? Otherwise it should be a check box or have a note available on the screen.

Naterater, 2018-12-02:
Even if you set the rates down (for example to 1hz) in mission planner, when you re-boot MP, it resets the rates back to defaults immediately on connect. That doesn’t make sense to anyone. To be honest, this should only be set in the parameter lists, and it’s confusing to have it both places and under different naming conventions.

meee1, 2018-12-02:
@Naterater pretty sure it doesn't 
only way I can see what you mention happening is if MP crash's on exit, and doesn't save the config.

Naterater, 2018-12-02:
@rmackay9 just described above how it does just by switching screens. The reboot does the same thing. Mission planner should never request changes unless users change it. Nobody should have to set those boxes to -1 in order for their previous changes to stay.

meee1, 2018-12-02:
This is a useability thing.
if MP never requested this then out of the box nothing would work. as there are no messages on by default.
I don't pan on changing the behaviour that is now here.
if you have a number set, MP will always attempt to request it.
if you have it set to -1, then its under user control.
there are still screens inside MP that do request this out of band however.
for example rc calibration.

Naterater, 2018-12-02:
Default telemetry rates on any firmware are not zero. Mission planner should not be changing these rates without user input to change them. It's not just requesting rates, it's overriding them to default every time. That overriding to default has to stop!

I understand that RC calibration has to have high rates; that makes sense why it should be temporarily overridden.

meee1, 2018-12-02:
"Default telemetry rates on any firmware are not zero" - please check your facts. they are infact 0.
[image]
connecting to a 100% clean copter.

closed 2018-12-02 by meee1