ArduPilot/MissionPlanner #2010: REQUEST_DATA_STREAM sent every 30sec, can't be disabled?

rmackay9, 2018-11-29:
I'm working on improving the performance of ROS/Cartographer by attempting to send the RAW_IMU message from the flight controller to the companion computer at high rates (100hz hopefully).

While working on this I've found that MP seems to be sending REQUEST_DATA_STREAM messages every 30 seconds.  This is probably good for most users but in my use case, this is stomping on the request from the companion computer to receive the raw sensor data at a higher rate.

Is there a way to switch off this REQUEST_DATA_STREAM?  maybe there should be able option in the drop-down to disable the request? I thought that "-1" would do it but it still seems to keep sending the requests.

meee1, 2018-11-29:
set the requested rate to -1 to not request it

rmackay9, 2018-11-29:
OK thanks, although as far as I can tell this doesn't seem to be working as expected then.
[image]

meee1, 2018-11-29:
was it only during the initial connect?

meee1, 2018-11-29:
ive double checked and to me it looks like its working. is the something else in the system that could be requesting it?  4 hz on everything seems odd, as its not a MP default

meee1, 2018-11-29:
MP does request the rates every 30 seconds. but only if the target rate is not being archived.
if its set to -1, the request is not sent at all. this could be related to the other issue you posted

rmackay9, 2018-11-30:
I can test again but the messages above (which are debug from the flight controller) showed that it was getting requests that look like they're from MP (I think these are MP's sysid and component id) exactly every 30sec no matter what I set the "Telemetry Rates" dropdowns to.  anyway, I'll test again with latest beta.

rmackay9, 2018-12-04:
I've retested with the latest beta and setting the telemetry rates to -1 does correctly turn off the requests to the autopilot to set the stream rates.  closing, thanks!

closed 2018-12-04 by rmackay9