ArduPilot/MissionPlanner #2012: LogViewer does misses to show CTUN items from .bin type logs

vierfuffzig, 2018-12-01:
log viewer doesn't show CTUN group when opening .bin type flashlogs. after conversion from .bin to .log the CTUN group is well accessible again.

Version 1.3.61 build 1.3.6909.22370

meee1, 2018-12-02:
could i get a copy of your bin?

vierfuffzig, 2018-12-02:
thanks for looking into this, michael!
just an example: https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/00000003.BIN
i suspected something wrong with the actual log i was trying to review first, but the issue shows in all log files actually.

cheers, basti.

meee1, 2018-12-02:
Thanks for reporting this. ive found the problem. and resolved it in the latest beta version

meee1, 2018-12-02:
the cause was that NTUN's internal id is 0. and i had code that ignored 0.

vierfuffzig, 2018-12-02:
awesome, thanks a lot!

closed 2018-12-02 by vierfuffzig