ArduPilot/MissionPlanner #2013: Zoom behavior and reset to home view

Naterater, 2018-12-01:
#### Issue details

The normal scroll zoom behavior in the flight and planning screens is to auto-center on the curser and move it to the center of the screen. This is unlike any other GIS application, and it's incredibly annoying having the curser jump across the screen when you only zoom one click.

I agree that the map should zoom centered on the curser, but do not move the curser; it just shows down future actions and disorients users that have to re-locate it.

#### Version

All

meee1, 2018-12-02:
i agree with the first point.

the second not so much. a user should be defining there home location. not doing so has other issues like distance to home.

Naterater, 2018-12-03:
Zoom also will not work when a waypoint or polygon point is hovered over. This is frustrating when you're trying to zoom in on a point because you have to move the cursor next to it in order to zoom.

meee1, 2018-12-03:
yeah, randy has an issue in regarding this. its on the todo list

aercamti, 2018-12-12:
I also have found that zooming is not intuitive, and wasted unnecessary time trying to find my area of operation again.

Having a (documented) shortcut to "zoom map (in or out) to complete flight plan" would be a helpful addition as well. 
(This would allow for a quick overview showing all waypoints, rally points, and home.)

"Clearing a mission shouldn't reset any views." - I agree.

skypuppy, 2018-12-25:
In Mission Planner, I set a mission fairly near my home, and I think I saved it. Then, after about a half hour of searching visually, I tried to set another mission near my grandparent's property about 150 miles away. However, when I select 'flight planner' for a survey grid near their property, it jumps back to the one already completed near my home. How do I get it to make a new flight plan near the grandparents so I can save it, too?  
Also, would be very nice if we could toggle on/off street names, to make it MUCH easier to find distant areas (like my grandparents place.)