ArduPilot/MissionPlanner #2018: Takeoff Mission Item Issues

Naterater, 2018-12-03:
#### Issue details
The takeoff mission item does not contain a header where the normal pitch angle column should be. This is an issue since the last beta update that I did today.

[image]

When adding takeoff and land WP's, the takeoff pitch angle and altitude cannot be changed. I always use 30-degrees as my takeoff angle and make the altitude the same as the survey waypoints. These "parameters" of the mission item should be an option in the menu that can be saved - just like all of the survey parameters that save for each mission. I've had numerous crashes and close-calls because the takeoff line was left at the default settings and not re-checked. This would prevent that.

Finally, there should be a pop-up box that warns users before writing a takeoff mission item with anything less than 10 degrees to a plane. To get around the issue above with the wrong parameters in the takeoff mission line, sometimes I add the takeoff line manually. Changing a mission line to takeoff changes its altitude but NOT its angle. That causes users to miss the 0-degree takeoff angle that causes crashes immediately.

#### Version
1.3.61 build 1.3.6910.18181

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2018-12-03:
where you connected at the time?
the display changes based on the vehicle connected.

because of the user numbers on copter vs plane, the default, non connected state is for copter.

meee1, 2018-12-03:
it used to be. but this caused other user issues. hence its no longer visable.

if you click the blank spot next to the baud rate, a new box will apear.

Naterater, 2018-12-03:
That box gives me no choices. I have a desktop that I only use for flight planning and never intend to connect a vehicle to it.

meee1, 2018-12-03:
[image]

Naterater, 2018-12-03:
I do not understand hiding boxes like that at all.