ArduPilot/MissionPlanner #2029: VFR HUD Compass is nearly visually useless

Naterater, 2018-12-06:
#### Issue details
[image]

The VFR HUD is supposed to be a quick-reference for critical in-flight data. The scrolling bar of numbers on top is very difficult to quickly interpret. That bar should be removed, and if heading data is so critical that it needs to be replaced, it should appear as something that comes from a proven instrument -- the directional gyroscope ("DG").

[image]

There's a reason just about every manned aircraft has one of these that looks the same. There's no need to re-invent the wheel here. Just use a similar concept. Even something looking roughly like a DG would be better than a scrolling bar of numbers at the top.

#### Version
All

#### Platform
[X] All

Naterater, 2018-12-06:
Wait a minute -- it looks like this feature already exists, at least partially! Those other gauges are already adequately represented on the VFR HUD, so I would suggest just modifying this one to work.

[image]

proficnc, 2018-12-06:
You will find the heading bar on most EFIS displays in modern aircraft exactly as it is on mission planner.

May I suggest engaging in some conversation about the list of issues you are raising? Discuss.ardupilot.org May be a better place

Naterater, 2018-12-06:
I've flown many EFIS systems in manned aircraft, and I have yet to see one that has a scrolling bar at the top. Here are systems from 3 major avionics suppliers, Garmin, Rockwell Collins, and Avidyne.

[image]

[image]

[image]

I think there has been plenty of discussion about these issues, and I'm breaking it down into specific tasks that can make things better from a UI perspective. Have a look at the thread I started almost a year ago. Maybe you would have some good insight to provide there. https://discuss.ardupilot.org/t/mission-planner-makeover/25394

meee1, 2018-12-06:
https://buy.garmin.com/en-AU/AU/p/514383

Naterater, 2018-12-06:
Now I have seen one, thank you. I still think that most aircraft, even with a heading indicator like that, will still have some form of rotating instrument like a directional gyro or a PFD with the integrated nav/track on it. We have the screen area to make it like a PFD, so I think it should be like a PFD. Down the road, the navigation lines (cross-track error) capability could be added too.

proficnc, 2018-12-06:
You set your EFIS up in a very cluttered manner! 

Many of us are licences pilots and avionics techs, I have been working on and flying full sized aircraft from the A-PC3 Orion down to the Jabiru -160... and even had the privilege of flying captains seat of a 737... having the heading bug on the primary EFIS or the HUD at the top of the screen is the primary case.  It’s even that way the Harrier that we have been restoring... (all be it a green overlay hud) 

Again though, this is a topic for a forum discussion, not the issues list. 

And again, if you really want to see some code changes to allow different themes.. PR’s are welcome.

Please also consider that Mission planner is the defacto industry standard, with now millions of users. Any changes here must be VERY well planned, and must have very solid reasoning behind them.

jmachuca77, 2018-12-06:
Another example

Naterater, 2018-12-06:
> You set your EFIS up in a very cluttered manner!
> 
> Many of us are licenced pilots and avionics techs, I have been working on and flying full sized aircraft from the A-PC3 Orion down to the Jabiru -160... and even had the privilege of flying captains seat of a 737... having the heading bug on the primary EFIS or the HUD at the top of the screen is the primary case. It’s even that way the Harrier that we have been restoring... (all be it a green overlay hud)
> 
> Again though, this is a topic for a forum discussion, not the issues list.
> 
> And again, if you really want to see some code changes to allow different themes.. PR’s are welcome.
> 
> Please also consider that Mission planner is the defacto industry standard, with now millions of users. Any changes here must be VERY well planned, and must have very solid reasoning behind them.

I kindly ask that you not attack the way I would set up an EFIS. The reason that the compass is most often shown below the artificial horizon is because of a pilot's traditional scan in the majority of aircraft, both digital and analog. The DG/Compass being directly below the artificial horizon is part of the standardized cross-checking of instruments that most pilots are familiar with. It's the same reason that airspeed is on the left and altitude is on the right - a carryover from analog instruments in the "six-pack."

[image]

I understand that there are a few systems that display a tape display style for the compass, but the majority of systems do not operate this way, and that's why I'm requesting the change. This is an issue to me, not a discussion point.

I strongly disagree that Mission Planner is the industry standard, and millions of downloads doesn't mean millions of users. I am doing my best to suggest things that will improve its usability and familiarity to professional users, and this is an item that I think would do a lot to make Mission Planner better. 

Changes are made to Mission Planner daily, and I don't think that this is a change that will hurt any user's experience. I am happy to test it, and I think there is already solid reasoning for a change.

The last two points are probably more discussion points than related to this topic, and I would be happy to comment them on a discussion thread if there is one related to the topic. If there is not, please feel free to start one and I will provide input.