ArduPilot/MissionPlanner #2030: VFR HUD should be blue over brown - not green

Naterater, 2018-12-06:
#### Issue details
Virtually every artificial horizon indicator uses blue and brown for contrast, not green. It's an industry standard, and it's proven. The VFR HUD should follow the same standard.

[image]

[image]

#### Version
All

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

lorbass, 2018-12-19:
Indeed, well observed.

And the altitude gauge should be adjustable as the speed gauge. Who climbs on 1000 m?

hdtechk, 2018-12-23:
Who cares what color it is your going to right click on it and turn your video on anyway. Then you see the real horizon