ArduPilot/MissionPlanner #2036: Install Firmware: add AC3.6.0, AC3.6.1, AC3.6.2

rmackay9, 2018-12-10:
It would be good if we could get some of the previous versions of Copter-3.6.x available in the Initial Setup / Install Firmware screen's "Previous firmware" links.  I don't actually think there are many good reasons to go back to previous versions.. all the later point versions are improvements but sometimes users think that something has been broken (in the software) when it's actually a hardware issue.  In these cases it is useful for users to go back to the earlier working firmware (and see it's still not working) so they can prove to themselves that it's not a software issue.

meee1, 2018-12-10:
I will update the list and push a new MP. tonight

rmackay9, 2018-12-24:
3.6.1 and 3.6.2 are available now in the MP, thanks!

closed 2018-12-24 by rmackay9