ArduPilot/MissionPlanner #2039: URI missing when uploading a new arduplane software

JeyPi, 2018-12-12:
https://www.goodluckbuy.com/minipx4-beta-version-flight-controller-support-pwm-ouptut-for-fpv-multicopter-quad.html

[image]

_Originally posted by @JeyPi in https://github.com/ArduPilot/MissionPlanner/issues/1915#issuecomment-446612699_

meee1, 2018-12-12:
are you sure the version you are trying to upload was available for the board you are using. I doubt it was

JeyPi, 2018-12-12:
You might be correct as the Pixhawk light was not having any problems with the new version.
The board I mentioned there fore seems to be defective or not compatible with the version. It was not compatible with any of the older versions either.

closed 2018-12-12 by JeyPi