ArduPilot/MissionPlanner #2040: Error when adding MAV_CMD_CONDITION_CHANGE_ALT - Upload wps failed - UNSUPPORTED

aercamti, 2018-12-12:
#### Issue details
When creating a waypoint plan and adding a CONDITION_CHANGE_ALT command following a WAYPOINT command, Mission Planner shows this error (following an attempt to "Write WPs"):

"Upload wps failed CONDITION_CHANGE_ALT
MAV_MISSION_UNSUPPORTED"

[image]

Whereas the Ardupilot Wiki states lists this as supported: http://ardupilot.org/copter/docs/common-mavlink-mission-command-messages-mav_cmd.html#commands-supported-by-copter 

(I am making a flight plan which needs the copter to descend upon arrival at the waypoint, and following an action, ascend again before moving on to the subsequent waypoint.)

Expected behavior: 

#### Version
1.3.62 build 1.3.6917.15581

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
hexa

#### Hardware type
SITL V3.7.0-dev

meee1, 2018-12-12:
it appears support was removed
https://github.com/ArduPilot/ardupilot/commit/fff21a1db98f915eee38bd698736951d21c1c277

aercamti, 2018-12-13:
Hmmh, that's unfortunate, if support of this regular MAVLink command has really been dropped... 
I guess, then it should also be deprecated from the drop-down list of available MAVLink commands in Mission Planner.

To achieve the behavior needed, this would now mean - each time - one would need to create two additional waypoints with identical lat/lon at each point to be flown to, which only differ in altitude, i.e. 

- wp1=arrive
- wp2=descend
- [action/ task]
- wp3=ascend; 
- then move on to next wp

This makes it really tedious to move a wp, as coordinates of 3 wps have to be manually re-matched in the table....

aercamti, 2018-12-13:
I _had_ looked at the link by meee1 - but don't have enough programming understanding to come to the conclusion you gave, thanks for the explanation @rmackay9 .

@meee1 If LOITER_TO_ALT is now supported, I guess the Command drop-down list in the flight planning section should be amended, and CONDITION_CHANGE_ALT should be replaced with LOITER_TO_ALT). Thanks!

rmackay9, 2018-12-13:
Yes, we should add LOITER_TO_ALT although it's only in Copter-3.7 and we weren't planning to even start beta testing for a few months..