ArduPilot/MissionPlanner #2044: loading PH4-mini firmware

tridge, 2018-12-17:
The PH4-mini firmware (Pixhawk4-mini) is not loaded properly with MP. The problem is that the PH4-mini and the Pixhawk4 both have the same fmuv5 bootloader, so there is no way for the GCS to distinguish the two. We need to use a separate firmware for the two of them, as PH4-mini has no IOMCU.
Ideally we need to display a list of matching firmwares when this happens, and let the user choose.
Right now what happens is that the Pixhawk4 firmware gets loaded, and when it boots it gets stuck with an error trying to update the IOMCU firmware.

meee1, 2018-12-17:
what vid and pid is being reported?
also is it using the new bootloader or the older one? ie is the bus reported description "fmuv5-bl" or "PX4 FMU v5.x"

tridge, 2018-12-20:
bootloader is this one:
[62494.810502] usb 2-2.1.1: new full-speed USB device number 63 using xhci_hcd
[62494.913067] usb 2-2.1.1: New USB device found, idVendor=26ac, idProduct=0032, bcdDevice= 1.01
[62494.913071] usb 2-2.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[62494.913074] usb 2-2.1.1: Product: PX4 BL FMU v5.x
[62494.913076] usb 2-2.1.1: Manufacturer: 3D Robotics
[62494.913077] usb 2-2.1.1: SerialNumber: 0
[62494.916552] cdc_acm 2-2.1.1:1.0: ttyACM1: USB ACM device

there is no info in the bootloader that can distinguish this board from the Pixhawk4 full board

tridge, 2018-12-20:
for comparison, here it the bootloader for a full Pixhawk4:
[62562.570739] usb 2-2.1.1: USB disconnect, device number 64
[62570.586408] usb 2-2.1.1: new full-speed USB device number 65 using xhci_hcd
[62570.688987] usb 2-2.1.1: New USB device found, idVendor=26ac, idProduct=0032, bcdDevice= 1.01
[62570.688990] usb 2-2.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[62570.688993] usb 2-2.1.1: Product: PX4 BL FMU v5.x
[62570.688994] usb 2-2.1.1: Manufacturer: 3D Robotics
[62570.688996] usb 2-2.1.1: SerialNumber: 0
[62570.690858] cdc_acm 2-2.1.1:1.0: ttyACM1: USB ACM device

as you can see, VID, PID and string are the same

tridge, 2019-01-21:
@jamming yes, I know we can detect it at runtime, the problem is that there is nothing that can be done to detect it until after the right firmware is loaded. The bootloader doesn't present this information to the GCS, so when users first load ArduPilot the GCS has no way to know which firmware to load.
I can see 3 possible fixes:
 - ship with the ArduPilot bootloader, which does distinguish between the boards
 - add a selection list to MissionPlanner when multiple firmware are possible and let the user choose
 - change the px4 bootloader to use a different board ID for the Pixhawk4-mini 

The simplest way fix is to start shipping with the ArduPilot bootloader. The ArduPilot bootloader does distinguish between the boards properly, and a GCS can select the right firmware every time. You may need to work with Don Gagne to allow px4 to select the right firmware if users choose to load px4. That should not be difficult.
The bootloaders are here:
https://github.com/ArduPilot/ardupilot/blob/master/Tools/bootloaders/Pixhawk4_bl.bin
https://github.com/ArduPilot/ardupilot/blob/master/Tools/bootloaders/PH4-mini_bl.bin
Alternatively you could modify the old px4 bootloader to use different board IDs for the different boards.