ArduPilot/MissionPlanner #2049: has no "Reference Paths" in "Properties"

IchMarcel, 2018-12-21:
#### Issue details
i want to build the MissionPlanner with VS2017, so i follow the introduction on the http://ardupilot.org/dev/docs/buildin-mission-planner.html,

but at this step",In the Solution Explorer right click the MissionPlanner project, Properties, Reference Paths", i cannot find the "Reference Paths" in the "Properties". So when i build the mission, there are so many errors, most of the errors are missing dlls.

i want to konw how to solve the problem...

THANKS!

#### Version
the latest version

#### Platform
[0 ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

n1kitap, 2018-12-24:
Hello, docs are outdated, there is no need to setup reference paths. You need to have Win10, latest vs2017, net core 2.0 sdk, net framework 4.6.1 sdk.

IchMarcel, 2018-12-25:
> Hello, docs are outdated, there is no need to setup reference paths. You need to have Win10, latest vs2017, net core 2.0 sdk, net framework 4.6.1 sdk.

so Win7 can not work properly? The OS must be Win10?