ArduPilot/MissionPlanner #2054: MAVLINK BatteryStatus seems to be handle incorrectly

olliw42, 2018-12-23:
#### Issue details

I think that MP doesn't handle the BATTERY_STATUS mavlink messages coming from different battery monitors, as identified by the id field, correctly.

I do have a setup which uses two battery monitors. The first, BATT, has cell voltages, while the second, BAT2, does not have that. In the MP status window I do see all battery_cell entries shown to be zero. The Mavlink Inspector shows that BATTERY_STATUS messages are send at 4.0Hz, and BATTERY2 messages at 2.0Hz. The voltages in the BATTERY_STATUS message all read 65536 (corresponding to invalid/not available), and the id field reads 1.

If in contrast I disable the 2nd battery monitor, so that only BATT with the cell voltages is active, I do see the battery_cell entries showing the expected values. Also the Mavlink Inspector shows that BATTERY_STATUS messages are send, now at 2.0Hz, and no BATTERY2 messages are present. The voltages in the BATTERY_STATUS message show the expected voltages, and the id field reads 0.

If I modify the ArduPilot code such that it only sends the BATTERY_STATUS message for the primary battery monitor, and not for all available battery monitors as it is doing, and have again both battery monitors enabled, then I do see the battery_cell entries showing the expected values; the Mavlink Inspector shows that BATTERY_STATUS messages are send, now at 2.0Hz, and BATTERY2 messages at also 2.0Hz. The voltages in the BATTERY_STATUS message show the expected voltages, and the id field reads 0.

To me it looks that the id field is not obeyed, and thus the data of the second battery monitor with id=1 overwrites or is taken as that of the primary battery monitor with id=0.

#### Version
I'm currently using MP 1.3.61, but have checked the release notes, and could find any which could fit that issue

#### Platform
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
irrelevant IMHO

#### Hardware type
irrelevant IMHO

#### Logs
hm
easy to reproduce IMHO

olliw42, 2018-12-23:
btw, a minor detail but could be fixed when at it:
then there are cell voltages present, higher voltages with 65536 value are shown in the Status window with 655536 ... I guess one would want to show them as 0, like it is done when all voltages are 65536

meee1, 2018-12-30:
do you have a tlog when both are connected?

meee1, 2018-12-30:
code is here
https://github.com/ArduPilot/MissionPlanner/blob/master/ExtLibs/ArduPilot/CurrentState.cs#L2185

I do check the ID's and only populate values when valid for the sensor id.

anther battery message could be interfering however

olliw42, 2018-12-30:
thx for your answer
I'm not used to tlogs ... I might find some time to get into them again

I'm very confident that ArduPilot sends out the BATTERY_STATUS for both battery monitors, because
(1) that's what it's code says
(2) when I tweak that code to only send one (for the primary monitor) it works (3rd point in the above)
I think the tlog would show that. The MP's Mavlink Inspector only shows the data for BATTERY_STATUS with id=1, but I think this is simply because both are coming in in very short time, so that only the last ist always seen. 

but I also see your code ... 

maybe my interpretation of the origin of the issue was too simple (it seemed so obvious from the evidences) ... maybe I will find some time to dig deeper into it (and get a tlog)

thy for all your work on MP !!!

meee1, 2018-12-30:
in  the latest beta MP you can graph in Mavlink Inspector.
this should show it flicking values if it is.

olliw42, 2018-12-30:
so, here some tlogs. Took a bit time because I had to redo and recompile the code.

tlog #1: www.olliw.eu/drop/michael-mp/2018-12-30_13-44-06-logwithcodechangedtosendonlyonebattstat.tlog

tlog #2: www.olliw.eu/drop/michael-mp/2018-12-30_21-29-04.tlog

tlog#1 is done with the ArduPilot code corrected such that it only sends the BATTERY_STATUS for the primary battery monitor. But it has two battery monitors enabled. In this case I did see MP displaying the expected battery cell voltages. 

tlog#2 is done with the original ArduPilot code for the BATTERY_STATUS handling, where it sends a BATTERY_STATUS for each monitor, which have id=0 and id=1. In this case I did NOT see MP displaying the expected battery cell voltages, it was showing all zeros.

However, now the weird, or may by hintful, observation, I found that when I replay the tlog#2 that then MP DOES show the correct battery cell voltages !!!

I'm not sure how I can prove to you that I don't see them in "real action". I thus did a screenshot of what I'm seeing in live :
www.olliw.eu/drop/michael-mp/michael-20183012-batterystatus-mp-screenshot.jpg
Note that all battery cell voltages are zero, but also note that the BATTERY_STATUS rate is 4Hz, not 2Hz (it seems that in the screenshot it momentarily fell below 4.0Hz, but this is really just momentarily, but you can note that BATTERY2 is just half the rate)

I maintain that something is up here, but not sure this helps identifying the issue's origin

olliw42, 2018-12-31:
just quickly tested with MP beta 1.3.6939.3924 and can confirm that the changes in the PR "CurrentState: 0 unused battery cells #2054" appear to work nicely. THX!

the main issue of the batterycells all showing zeros in the mentioned situation remains however

olliw42, 2019-01-12:
had a chance to test latest beta MP, and the PR seems to fix the issue.
thus closing it.
:)

closed 2019-01-12 by olliw42