ArduPilot/MissionPlanner #2060: MAVLINK Component: Param Fetch Hangup

olliw42, 2018-12-28:
#### Issue details
I have a setup with a Pixracer and a MAVLINK component attached to one of it's serial ports. Param fetches for the MAVLINK component are dodgy.

When reading the parameters of the MAVLINK component for the "first time" (see below), the read hangs up: It reads the parameters until the last was received (as visible from the dialog box), waits, and waits, and then displays "Failed to update home location (2)". After that the read must be aborted manually.

As regards then this happens: This does not happen for follow up reads, i.e., the systematics appears to be such: It happens, I abort manually, and any follow up "Refresh Params" works nicely. If I however e.g. switch to the QUADROTOR, and then back to the MAVLINK component, and do "Refresh Params", the hangup happens. After that, again all "Refresh Params" would work. And so on. The hangup also happens when connecting MP to the pixracer (as MP then obviously tries to read the params of also the component).

This happens independent on wether the MAVLINK component uses Mavlink1 or Mavlink2.

To me it seems that MP expects to receive a particular parameter/value even for MAVLINK components, which it shouldn't do since MAVLINK components might not have that parameter.

#### Version
MP 1.3.62

#### Platform
IMHO irrelevant
#### Airframe type
IMHO irrelevant
#### Hardware type
IMHO irrelevant
#### Logs
hm, easy to reproduce

meee1, 2018-12-30:
could you try the latest beta MP? and provide a tlog?

olliw42, 2018-12-30:
I have seen that you just have merged a possible solution to the MAVLINK component hand up. 

How long would this take it to get into the latest beta release of MP?

I've mentioned this in the BatteryMonitor issue, when you are at it you also could change this
https://github.com/ArduPilot/MissionPlanner/blob/master/ExtLibs/ArduPilot/CurrentState.cs#L2185
to 

```
if (bats.voltages[0] != ushort.MaxValue)
 {
  battery_cell1 = bats.voltages[0] / 1000.0;
  if (bats.voltages[1] != ushort.MaxValue) battery_cell2 = bats.voltages[1] / 1000.0; else battery_cell2 = 0.0;
  if (bats.voltages[2] != ushort.MaxValue) battery_cell3 = bats.voltages[2] / 1000.0; else battery_cell3 = 0.0;
  if (bats.voltages[3] != ushort.MaxValue) battery_cell4 = bats.voltages[3] / 1000.0; else battery_cell4 = 0.0;
  if (bats.voltages[4] != ushort.MaxValue) battery_cell5 = bats.voltages[4] / 1000.0; else battery_cell5 = 0.0;
  if (bats.voltages[5] != ushort.MaxValue) battery_cell6 = bats.voltages[5] / 1000.0; else battery_cell6 = 0.0;
}
```
or similar. I could then test this too once the new beta comes. And I think it is an obvious reasonable change. :)

THX!

meee1, 2018-12-30:
beta is available now. its part of the CI

olliw42, 2018-12-31:
just quickly tested with MP beta 1.3.6939.3924 and can confirm that the changes in the PR "MainV2: dont request home for invalid autopilot #2060" appear to work nicely. THX!

I close it therefore.

closed 2018-12-31 by olliw42