ArduPilot/MissionPlanner #2061: Kind request to add UAVCAN_NODE_INFO support

olliw42, 2018-12-30:
Hey Michael,

I'm working on a UAVCAN-MAVLink bridge, and have a first well-working prototype. Gladly, your MissionPlanner already supports UAVCAN_NODE_STATUS, and is working really nicely for this. For a demo picture see http://www.olliw.eu/uploads/uc4h-mavlinkbridge-mp-gui-v01.jpg (a short write up is [here](http://www.olliw.eu/2018/uavcan-for-hobbyists-blog/#20181230)).

What however seems to be missing in MissionPlanner is support of the UAVCAN_NODE_INFO mavlink message. A description is here: https://mavlink.io/en/guide/uavcan_interaction.html

I herewith kindly want to encourage you to add this feature. 

(1) In it's most basic form it only would read off the UAVCAN node's name and display it, such that it's easier for the user to realize which node is which. Currently, with just all the numbers it's not really convenient, as you may confirm yourself by looking at the picture I have linked too in the above. An obvious place to display the node name would be in the drop down list of all components, behind the ID, similar to the QUADROTOR.

It would be cool however if the support could go beyond that. 

(2) For instance one could display the node name also on top of the parameter page, so that it is easier for the user to see to which node the parameters belong to. There would be then also space to show additional information on the node ... health, mode, version, vendor status code, .... whatever is considered useful.

(3) Another nice feature would be if the drop down list could be sorted in some useful way, e.g. such that all UAVACN nodes appear with increasing NodeId, or sorted by name (which maybe is more useful when dynamic node ID allocation is used), and maybe flight vehicles first and then all related UAVCAN nodes following it.

(4) Along similar lines I would find it very helpful if the parameters were NOT sorted alphabetically, but were shown sorted indeed by their index. Parameters of UAVCAN nodes do not usually/necessarily have the prefix like the MAVLink parameters, by which they are organized into "chapters", but are arranged by the UAVCAN node builder into a somewhat proper sequence, since tools such as e.g. the UAVCAN GUI Tool just list them sequentially by index, and a UAVCAN node builder presumably want them to show up nicely there.

(5) Also, the MAV_CMD_UAVCAN_GET_NODE_INFO could be supported, but I think that's not absolutely necessary, since one could also agree on sending the UAVCAN_NODE_INFO unconditionally (it's a big packet though).

(6) Finally, and I know that this is a bit of a digress from MissionPlanner principles, it would be great if there would not be only read and write buttons, but also a STORE button, with which one would initiate that parameter changes are stored permanently in the node. I know that MP sends the PREFLIGHT_STORAGE mavlink message some 10 secs or so after a parameter change to facilitate this, and it could be used here too, but the point is that it's not similar to how e.g. the UAVCAN GUI Tool works, which may confuse users, and in fact can result in some subtleties which are hard to understand by the user.

Lots of wishes, I know, and Christmas just has passed, I know, but ... :)

THX for all your work. Olli

meee1, 2018-12-31:
(5) often ArduPilot sends this type of packet when the parameter list is requested. ie once off send with the parameters

olliw42, 2018-12-31:
just quickly tested with MP beta 1.3.6939.3924 and the changes in the PR "MAVLinkInterface: add UAVCAN_NODE_INFO name support #2061" make it nearly to work:

the drop down field needs to be updated too !!

upon connect, MP initially will not know the name because it only has seen NODE_STATUS, a NODE_INFO will come in only later, but the new name will not be shown in the drop down field until the user decides to click on it to unroll it, only then it will become visible.

a small but user request: The UAVCAN node names tend to be (much) longer than the MAVLINK names (the uavcan node names can be 80 chars or so). With the current width of the drop down list only a small fragment is seen. Currently I can see only the first 19 characters of its name.
It would be thus cool if it could be significantly wider. Since there is plenty of space on the right it easily could be wider, I think.
Also the drop down field itself could be wider, it only shows me 14 chars of the name. It for instance could have the same width as the COM port and BAUDRATE field together.

THX so far !!

olliw42, 2019-01-01:
hey

so, I've spend a long day yesterday and also today with working out a "best procedure". I want to share in the unlikely case it's of interest. I don't want to bother you with the failed attempts I made, just mention that the procedure of sending the  UAVCAN_NODE_INFO with each PARAM_REQUEST_LIST did not work really well, in contrast to what I implied before.

Two procedures I found to work best. The key is to ensure that MP gets the UAVCAN_NODE_INFO messages when it is connecting, before it has finished connecting, i.e. finished downloading the parameters. 
* One approach is to simply send the UAVCAN_NODE_INFO messages unconditionally. Every 15 secs I found to work well for me. Since downloading all the ArduPilot parameters easily takes longer than this, this is sufficient to ensure that MP has received the UAVCAN_NODE_INFO of all currently known UAVCAN nodes before it is done.
* As the second approach I modified the ArduPilot code such that it sends out a MAV_CMD_UAVCAN_GET_NODE_INFO to all known MAVLINK channels when it detects that MP wants to connect (it isn't doing anything with this message currently). The MAVLINk bridge responds with sending the UAVCAN_NODE_INFO of all currently known UAVCAN nodes. 

So, both approaches achieve the main goal of informing MP before it has finished connecting. Once that is achieved, it all really works smoothly and fine. Even the update issue I mentioned before isn't an issue anymore. With either of these procedures I couldn't yet create a situation where it would not behave as I would expect it to behave. I am completely satisfied so far.
The advantage of the first approach is it's simplicity, but it somewhat congests the channels since the UAVCAN_NODE_INFO packets aren't small. This is nicely avoided by the 2nd approach but has the disadvantage that the flight controller firmware needs to be modified.
I actually decided to use a combination of both methods, i.e. initially the bridge uses the unconditional send out approach, but once it has received a MAV_CMD_UAVCAN_GET_NODE_INFO message it will disable the unconditional sending and kind of switch over to the 2nd method. That way it will work nicely independent on whether MAV_CMD_UAVCAN_GET_NODE_INFO is supported by the FC or the GCS.

With these findings I wish to update my "list of wishes or suggestions"

* (1) In contrast to my earlier statement I think it would be great if MissionPlanner could support MAV_CMD_UAVCAN_GET_NODE_INFO, i.e. send it when it wants to establish a connection. It sure works equally well if this is done by the flight controller, I however find that illogical, i.e. to let the flight controller figure out that a GCS wants to connect and then to trigger sending the GCS the information it wants. To me it would be just more natural and logical if it is the GCS which triggers getting the info it wants.
* (2) I would really wish that the parameters would be displayed by index and not sorted by alphabetically for the UAVCAN nodes.
* (3) I would really wish that the width of the drop box is wider, so that more of the UAVCAN names is seen, and if that list would sorted in some reasonable ways, e.g. flight controller first followed by its nodes sorted alphabetically by name, and in case of equal names by node id (component id). I think that both could be achieved easily.

The other wishes remain, but these would be the 3 wishes I really would love to see.

Thx for the addition of UAVCAN_NODE_INFO, it's really an improvement, and all the best, 
Olli

BTW: I'm not sure if that is a feature or a bug, but then a MAVLINK component or UAVCAN node disappears, the drop box still will show them. I would find it more logical that all components/nodes which have become inactive, i.e. haven't send a HEARTBEAT or a  UAVCAN_NODE_STATUS for a couple of seconds, would also disappear, like it is with the flight controller, and like it is with the UAVCAN specification. This also would e.g. help preventing to do e.g. ParamRefreshes for inactive components/nodes .

olliw42, 2019-01-02:
many thx for the quick PR.

I have not tested it but just looked at the code, and I think that - if I'm understanding it correctly - this isn't going to work well: The UAVCAN_NODE_STATUS is being send out quite frequently, it is so to say the HEARTBEAT analog, typically every 1 sec or so. With your change a MAV_CMD.UAVCAN_GET_NODE_INFO would be send out every 1 sec, which would make the MAVLINK bridge to send out the UAVCAN_NODE_INFO for ALL known nodes every 1 sec ... this would really kill bandwidth for no reason. (for 8 nodes like on my copter this would be 8x(116 + packet header stuff) bytes per sec)(bytes not bits)(or ca 15% on a 57600 channel)

I have written down my findings&conclusion for a reason :).

olliw42, 2019-01-02:
as a little motivation, if it needed any: https://youtu.be/Y60CY4Yoc9A

olliw42, 2019-01-10:
as regards your PR https://github.com/ArduPilot/MissionPlanner/commit/2d5ca6fba53492a44840eccecd4435d97c6faf4c you certainly have realized that I missed the effect of the line `if (!MAVlist.Contains(sysid, compid, false))`. So, I'll take back my comment that it ends in a catastrophe.

BUT: You nevertheless want to change the code, IMHO.

To explain, let's again use the example of my copter in it's current config with 8 attached nodes, so there are in total 9, since there is also the pixhawk. These typically send their UAVCAN_NODE_STATUS messages in intervals of 1 Hz. That is, upon connect, from the first call of  `if (msgid == (uint) MAVLINK_MSG_ID.UAVCAN_NODE_STATUS)` to the last call it will take about 1 secs. That is, within this 1 sec, MP will receive 9x a UAVCAN_NODE_STATUS and send out 9x the UAVCAN_GET_NODE_INFO, and the mavlinkbridge is supposed to answer to each of them with sending out the UAVCAN_NODE_INFO for all nodes known to it, which is 9. Thus, within 1 sec you will get traffic of 9 x 9 x (116 + packet header stuff) bytes, which exceeds the bandwidth of a 57600 channel by nearly a factor of 2 !!! So, you effectively block the channel for nearly 2 secs. 
This is btw what I do indeed see in experiment,.

What you want to do is IMHO that you limit the rate with which you send out the UAVCAN_GET_NODE_INFO message to e.g. 1Hz. That is, something along the lines of this
```
static bool send getnodeinfo = false;
static uintXX_t send getnodeinfo_last_ms = 0;
 if (msgid == (uint) MAVLINK_MSG_ID.UAVCAN_NODE_STATUS)
                   {
                        var cannode = message.ToStructure<mavlink_uavcan_node_status_t>();
                        // add a seen sysid
                        if (!MAVlist.Contains(sysid, compid, false))
                        {
                            // ensure its set from connect or log playback
                            MAVlist.Create(sysid, compid);
                            MAVlist[sysid, compid].aptype = MAV_TYPE.ONBOARD_CONTROLLER;
                            MAVlist[sysid, compid].apname = MAV_AUTOPILOT.INVALID;
                            MAVlist[sysid, compid].CANNode = true;
                            setAPType(sysid, compid);

                            // new device, so request node info
                            send_getnodeinfo = true;
                        }
}
now_ms = millis();
if( (now_ms - send_getnodeinfo_last_ms) > 1000){
    send_getnodeinfo_last_ms = now_ms;
    if( send_getnodeinfo  ){
         send_getnodeinfo = false;
         doCommand(sysid, compid, MAV_CMD.UAVCAN_GET_NODE_INFO, 0, 0, 0, 0, 0, 0, 0, false);
     }
}
```
This is untested of course, and probably isn't working for you, just written down to give the idea.

I have to say, currently I'm blocking out the UAVCAN_GET_NODE_INFO messages from MP, since it's not an improvement over what I have.

olliw42, 2019-05-15:
I think can be closed :)

closed 2019-05-15 by olliw42